computergeek1507/PB_16 #1: PB_16 Voltage Regular

libby2cm, 2020-06-17:
Hi, it looks like at one point this board used a buck converter, but now uses a LM7805.  Is that on purpose, or were there issues with the buck?  If the 7805 is right, does it need a heatsink or anything?

computergeek1507, 2020-06-17:
You should use a OKI-78SR-5 buck converter. The Kicad 3D renders show a 7805 because I have not made a 3D model for the OKI-78SR-5 buck converter. The BOM also has a  OKI-78SR-5 buck converter.

libby2cm, 2020-06-17:
Ah, got it.. missed it in BOM.  Just sending boards out to fab, so wanted to double check.. thanks!

closed 2020-06-17 by libby2cm

libby2cm, 2020-06-17:
Just a heads up, looks like it's in digikey library now, with 3d model (horizontal & vertical)

[image]

computergeek1507, 2020-06-17:
Do I also see the polifuse 3d model as well in your screenshot?

libby2cm, 2020-06-18:
> Do I also see the polifuse 3d model as well in your screenshot?

Yes, I didn't change anything from your last commit though, so new version of kicad must have included new libraries/models.