computergeek1507/PB_16 #2: Mosfet switch on ports

sslupsky, 2021-02-05:
Due to the safety issues with pixels, I wonder if it would be a good idea to put a mosfet on each port so they can be shut down?  The mosfets could have a common gate signal.  When the show is over, FPP could turn off the power to the pixels.

Kosh42, 2021-02-05:
I'd argue the "safety issue" is to do with cheapest price point pixels rather than quality pixels, but argument for another day.

In this case I believe that if you want to shut down the pixels side, but keep the controller live, you'd power it from aux with a 5V input, them just shut down the pixel PSU feeding the main port inputs.

It's how I plan to do it, from a safety and efficiency point of view.

sslupsky, 2021-02-05:
Agreed.  Race to the bottom and we are not in control of the design of the pixels so the suppliers make crazy mistakes.  It would be nice to control the power to the pixels from FPP.  This would provide people that want to have permanent lights a bit of a better safety margin.  It would also give Scott's design an additional feature the other controllers do not have.  If people are concerned about the cost of adding the extra fets, we could add some 0 ohm resistors to bypass them and allow people to build without the fets.

processor83, 2021-03-31:
im sure its possible alredy to control a relay with some GPIO to turn on or off the power supply source. If using hp power supply you can take adventage of the standby 5V and the enable pin... for meanwell power supply you can have a 5V power supply and control a contactor to turn on off the AC source or even control a smart plug so i dont see any need to make huge modification to the controller here.

sslupsky, 2021-04-01:
But that solution requires two power supplies.  One to run the board and another for the pixels.  The better solution is to have the control on the controller.  A TI eFuse costs less than $0.50, are UL/CSA listed and provides an electronically resettable output.  So, you can get rid of the thermal fuses and fuse holders and the cost of those components.

processor83, 2021-04-02:
Do you have the part number? How may gpio it would need to control all those fuses?

sslupsky, 2021-04-02:
TPS2595, TPS25947, TPS25980 are some parts I think would work for this application.  You could connect the EN pin of all the fuses together and use one GPIO to turn them on/off.  If you want to control them all individually, you would require one GPIO per fuse.  Or, use an I2C port expander and drive the EN pins from that and use the I2C bus.

sslupsky, 2021-04-02:
Note, the max current of the TPS2595 is 4A.  All pixel strings sold these days come with 22 AWG wire so the current limit of an output port should be limited to the wire capacity which is 3A for 22 AWG wire.  Thus the TPS2595 would be suitable.

ptr727, 2021-12-23:
FET's switching the DC output sounds like an interesting idea, I was considering relays or SSR's in front of the PSU's.
I'm contemplating building a DIN enclosure next season, AC breaker, 5V DIN PSU for controller, and multiple 10A (12V, 24V, 48V) DIN PSU's per light group, maybe 2 to 4 outputs per PSU, resettable fuses, I2C INA236 / ACS712 current monitoring, dedicated injection outputs running 24/48V DC-DC 12V at end of long runs.

processor83, 2022-10-15:
i made some test with HP server power supply they have 12 stanby power and i used a mini buck converter and feed the beaglebone from this all the time and a gpio at the begening of the schedule to turn on the power supply and turn off at the end of the show i used opto coupler to protect the gpio it work pretty good and need only one psu...