computergeek1507/PB_16 #3: Flash bin to ROM

Kosh42, 2021-02-22:
HI,

I've manager to create my own bin file for the BBB16 v2, but the PB16 files say to go to Help > About, then "Scroll to Cape/Hat Firmware Upgrade". I have no such option when I upload this to the FPP instance, only a section called "About Cape/Hat" on that page.

I know the bin is valid as placing in it /opt/fpp-vendor/cape-eeprom.bin works as expected. This is of course putting my own public key in /opt/fpp/scripts/keys/ to match the one used when generating the bin, as without that I can't even get to this far.

I have tried the old FPP way with a sudo cp to the eeprom but this again doesn't work. When I nano this though I do see it still reports all the "filler" data in the unused end of the EEPROM. 

Running `sudo journalctl -u fppcapedetect` gives me:

`-- Logs begin at Thu 2019-02-14 10:12:00 UTC, end at Mon 2021-02-22 20:45:54 UTC. --
Feb 14 10:12:03 FPP systemd[1]: Starting FPP Cape Detection...
Feb 14 10:12:03 FPP detect_cape[163]: Running detect_cape
Feb 14 10:12:04 FPP detect_cape[163]: Running fppcapedetect
Feb 14 10:12:08 FPP detect_cape[163]: Did not find 0x50 on i2c2, trying i2c1.
Feb 14 10:12:08 FPP detect_cape[163]: Did not find eeprom on i2c.
Feb 14 10:12:08 FPP detect_cape[163]: EEPROM file doesn't exist .
Feb 14 10:12:08 FPP detect_cape[163]: Using /opt/fpp-vendor/cape-eeprom.bin
Feb 14 10:12:08 FPP detect_cape[163]: Found cape BBB16, Version v2, Serial Number: 12345
Feb 14 10:12:08 FPP detect_cape[163]: Done fppcapedetect
Feb 14 10:12:09 FPP detect_cape[163]: Done
Feb 14 10:12:09 FPP systemd[1]: fppcapedetect.service: Succeeded.
Feb 14 10:12:09 FPP systemd[1]: Started FPP Cape Detection.`

When I try this on a fresh fpp without the public key I get:
`-- Logs begin at Thu 2019-02-14 10:12:00 UTC, end at Mon 2021-02-22 21:03:40 UTC. --
Feb 14 10:12:03 FPP systemd[1]: Starting FPP Cape Detection...
Feb 14 10:12:03 FPP detect_cape[168]: Running detect_cape
Feb 14 10:12:04 FPP detect_cape[168]: Running fppcapedetect
Feb 14 10:12:05 FPP detect_cape[168]: Did not find 0x50 on i2c2, trying i2c1.
Feb 14 10:12:05 FPP detect_cape[168]: Did not find eeprom on i2c.
Feb 14 10:12:05 FPP detect_cape[168]: EEPROM file doesn't exist .
Feb 14 10:12:05 FPP detect_cape[168]: Could not detect any cape
Feb 14 10:12:05 FPP detect_cape[168]: Done fppcapedetect
Feb 14 10:12:06 FPP detect_cape[168]: Done
Feb 14 10:12:06 FPP systemd[1]: fppcapedetect.service: Succeeded.
Feb 14 10:12:06 FPP systemd[1]: Started FPP Cape Detection.`

Any thoughts welcomed, and apologies if me trying to hash this together I've made some schoolboy error.

Many thanks,

Tim

computergeek1507, 2021-02-26:
Try flashing this file and let me know if it works
[BBB16-eeprom.zip](https://github.com/computergeek1507/PB_16/files/6047540/BBB16-eeprom.zip)

ukdel, 2021-03-11:
hi
I am doing  a similar thing to Tim and I did load your BBB16-EEprom file and seemed to work a treat (I still have to run it with full pixels connected). The only additional issue is that I am running one of your multi expansion board  from the main BBB16. It does allow for an expansion card, and includes for differential or standard ports, but not a mix of the two. Do you have a solution for this yet? Many Thanks
D

computergeek1507, 2021-03-12:
From a configuration standpoint the only difference between a differential or standard ports is the ability to setup smart receivers. For FPP the pixel data output is basically the same. For the multi expansion just use the differential expansion option. The first two groupings(8 ports) are the local standard ports and the last two groupings are the diff receiver ports.

Kosh42, 2021-03-13:
Many thanks for your work on this. Not sure where else to post this without bugging you, but any chance I can have a look at the files you created the bin from for learning purposes? I've created them for a different thing I'm working on and they are not quite working as expected (many the control buttons).

ukdel, 2021-03-13:
Hi kosh.. 

Scott produced the files.. and also produced the changes in xlight. He is a star.

I got it all working with hi s files.

Derek

..on the move

closed 2021-04-18 by computergeek1507