computergeek1507/PB_16 #4: Smart receiver documentation?

processor83, 2021-03-31:
Hello, i see you work on a smart receiver 3 in 1. Do you have any documentation you can share so i can learn how they send information about the receiver ID to the smart receiver. I have read FPP source code but its not 100% clear to me!  I want to try learning AVR and do it with attiny one my side.

Thank you for your help!

Kosh42, 2021-03-31:
Looking at a commercial differential board, I noted an ATTiny chip on it I think. As my coding is so bad, I gave up trying to reverse engineer it further from that point. Where did you see that Scott was working on a differential board?

processor83, 2021-03-31:
I also try to replicate those smart receiver you talk about.  When i see something i always want to learn how it work and build it! Very steap learning curve for me learning PCB design shematics  and now try to add prog to it all! Im far from anything working. What i know is that it can be done with Attiny416 and that it have to read the receiver id in the data stream and this will turn on and off a switch inside the attiny416 to allow data or not to the 4 pixels ports based of receiver ID.

I looked at falcon pi mplayer source files since it is used on kilp board and i kind of understand that is pull the data pin low to tell the attiny416 to listed for ID then send the id and then return to led data mode. It do this super fast to turn the switches on and off on the smart receivers.

Maybe it would be easy to see on a scope. I also asked Daniel Kulp about it since he is the one that found it and added this to fpp.

You can look at  smart_receiver_max folders in this repo for this project. Its exactly 3 bank of 4 so i guessed it must be compatible with falcon kulp but i also see a the smart_receiver_max_out so maybe he work on is own way of doing it and will break out 12 full ports over one signle cable.

processor83, 2021-03-31:
Ok i will write it there since i just thinked of something and i guess its would be the way they do it. strop led data on the 4 pins use those 4 data pins in high or low state to comunicate the id and resume led data it make sense for pi gpio pin high or low its like those 4 dip switches on the receiver ... so simple now thta i think about it  and im 80% sure its like this was thinking they where addind packed to the ws2811 stream but attiny416 is too slow to decode all this data real time.

processor83, 2021-03-31:
Ok afther making some more thinking its like this but each port are individual and the id is send  individually to each ports.

computergeek1507, 2021-04-01:
There is no real documentation, but here is a screenshot of the data from my logic analyzer.

[image]

processor83, 2021-04-01:
Thank you for this it will help us!
Can i assume this.
180ms low to tell the ic to listen for id.
180ms port number high to tell the id to the smart receiver
180ms low to return to pixel data mode.

Now to start to play with this info :)
Thank you again!

computergeek1507, 2021-04-01:
the data is not high for 180us it is toggling at 800khz.

processor83, 2021-04-01:
ok so its like they are sending 100% white if i unserstand correctly from the source code of fpp.

Thank you.

ninjinja, 2023-07-04:
Is there firmware available for the smart receiver designs, and are you willing to share the source code?   Thanks