computergeek1507/PB_16 #10: Pull Requests

sslupsky, 2021-12-04:
@computergeek1507 Are you taking pull requests?  I've made a SMT version of the BBB-16 and had JLCPCB assemble it.  The board can now be completely assembled by JLCPCB including the through hole parts.  The only part JLCPCB will not currently assemble are the two headers for the BBB because it must be mounted on the bottom side of the board.  JLCPCB support assembly on only one side.

computergeek1507, 2021-12-04:
My only request is that it's licenced under an open source hardware license, so myself or others can use it and modify it.

sslupsky, 2021-12-05:
Sure, that's not a problem.  Should I submit it as a commit to BBB 16 or rename it something like BBB 16-SMT?

computergeek1507, 2021-12-08:
BBB 16-SMT or BBB 16-SMD is fine with me

ptr727, 2021-12-23:
@sslupsky that sounds cool, can't wait to see it!

ptr727, 2022-01-20:
Hi @sslupsky are you still planning on a PR here, if not, would you share your project publicly?

computergeek1507, 2022-01-20:
I was wondering as well, if not I will start one

sslupsky, 2022-01-21:
@ptr727 Yes.  Apologies for taking a while.  I was doing some clean up, then KiCad updates and then got distracted with some sequencing.  I'll get back to this soon.

@computergeek1507 One design change I have been contemplating is to implement a switch on the port power outputs.  Is there a spare GPIO that can be used for this purpose?  If not, I could use an I2C port expander to add the additional GPIO.  The switches also have a current sense so it would also possible to monitor current on each port.  Do you think the current monitor would be useful?  Can these control elements be integrated into FPP?

ptr727, 2022-01-21:
+1 for output switch and current monitor (e.g. TI TPS).
I'd like to run electronics powered 24/7, but power up the LED power only when in use (permanent install safety and power savings).
I'd also like to experiment and drive output by current limit threshold vs. % power setting trying to get output max below fuse rating (resettable fuse?).
Don't know what V ratings for FET/switch you're looking at, but another thing I'd like to try is running output at 24V/48V (lots of newer datacenter gear running 48V DC, hoping it brings prices down for DIN mount PSU's), and then a 24V-48V DC -> 12V DC per string section.
Talking about PSU, rather than big single PSU, want to run n-outputs per DIN mount PSU, e.g. modular 2/4/8-outputs per PSU, and same outputs could be used for power injection still with current monitoring / limiting.
Anyway, lots of things one could do, just need time ;)

sslupsky, 2022-01-21:
The component that I was considering for this is the TPS2597xx.  It has the switch and built in current monitor as well as a current limited output.  So, we can eliminate the fuses.  Probably should add some LED's.  Recommended max input voltage is 23V for this part.  It rated up to 7A.

ptr727, 2022-01-21:
The TPS1HB08-Q1 is 8mR 40V, so safe for 24V, there are others in the automotive high side switch series.
https://www.ti.com/power-management/power-switches/high-side-switches/products.html

For 48V not so easy, even the INA3221 is 26V, and then need a FET, but it is a 3x channel with 4x selectable I2C address, so 12 outputs per I2C bus without a multiplexor.

But this is just browsing, no idea what is available at JLC for manufacturing.

sslupsky, 2022-01-24:
@computergeek1507 I am preparing the BBB_16_SMD PR and had a question about the organization of the repo.  I've been wondering if each board shoud tracked separately?  The way it is now, commits are interspersed among the individual boards (projects).  As I understand there are two approaches to doing so, using a submodule or a subtree.  Let me know what you think.

WaldoWood, 2022-02-09:
@computergeek1507 Pull requests? Hi, I've made a modified version of the BBB-16 to fit my own expansion boards, as I don't want to use Holiday Coro Flex Boards. Would it be ok to use your BB16-Flex-eeprom?

ptr727, 2022-05-07:
@sslupsky any progress on switchable outputs on a JLC buildable SMD board?

sslupsky, 2022-05-08:
The design is complete.  I am waiting for the switch to come in stock and then I should be able to order some prototypes from JLC.  You can switch the outputs in banks of 8 ports (so, two banks).  You can monitor the load current on any output.

Design enhancements include an RTC and battery, audio DAC, speaker amplifier and FM transmitter.  I also changed the DMX RJ45 to a diff port and moved the DMX port to the DMX terminal block which is now six pins.

Also, the power path is now jumperless and auto configures itself based on the power input connections.

sslupsky, 2022-05-08:
Here is a render of the pcb
blob:null/47971c57-c64c-4068-a306-0a56421b17fc

ptr727, 2022-07-16:
I see that ExperienceLights announced their new Genius controllers have auto resettable polyfuses and current monitoring, they posted a video demoing on the XLights FB group; using current monitoring to good use doing auto pixel counting and pixel fault testing.

sslupsky, 2022-07-16:
Yes, using the current monitor, you could detect stuck pixels and dead pixels.  I am still waiting for the parts to become available.  Should be soon.

ptr727, 2022-11-02:
@sslupsky any update on publishing your design?

kingofkristmas, 2022-12-23:
> Yes, using the current monitor, you could detect stuck pixels and dead pixels. I am still waiting for the parts to become available. Should be soon.

Excited and waiting to see what you come up with!

sslupsky, 2022-12-23:
I think I have found a substitute for one of the parts that will allow me to build a few of these.  I am looking into a couple of the other parts that have disappeared from distribution now.  Hopefully, early in the new year I will be able to build one.