computergeek1507/PB_16 #11: FPP reports abnormal condition:  Could not not initialize output type BBB48String

sslupsky, 2021-12-09:
I have an issue where FPP is reporting an abnormal condition.  I am using the BBB16 and I believe this problem began when I flashed the BBB16v2-eeprom.bin file but that might just be coincidental.  The eeprom appears to have been flashed but I did not notice a message indicating the flashing was complete when I performed the flash.

xLights appears to upload the output configuration.  I see the "FPP output upload complete" message in the bottom left corner.  Note however that xLights does not have an option for "BBB16v2".  The only option is "BBB16".

[image]

When I inspect the output configuration in FPP, it appears to be correct.

[image]

When I inspect fppd.log I can see a couple ChannelOut messages:

```
2021-12-09 10:13:49.669 (2365) [ChannelOut] /opt/fpp/src/channeloutput/BBB48String.cpp:293: No output pin configuration for BBB16
2021-12-09 10:13:49.671 (2365) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:395: Determined range needed 0 - 7
```

I notice the ChannelOut message refers to BBB16 but the output configuration in FPP shows the device as BBB16v2.  I am not sure if this is relevant or an indication that ChannelOut thinks this is an older version of the board.

Not sure if this is an FPP issue or related to the eeprom.

FPP version is 5.4.1

computergeek1507, 2021-12-09:
xLights is setting the wrong controller type in the FPP bbb-strings JSON file. I will have to update xLights

sslupsky, 2021-12-09:
Yeah, that is the problem.  Here is the relevant part of co-bbbStrings.json:

```
            "subType": "BBB16",
            "pinoutVersion": "1.x",
            "type": "BBB48String",
            "startChannel": 1,
            "channelCount": -1,
            "enabled": 1
```

I changed it to "BBB16v2" and the abnormal condition appears to be resolved.  The fppd.log file now shows that ChannelOutput is loading the output configuration properly:
```
2021-12-09 11:05:23.830 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:356: BBB48String 0:  Determined range needed 30240 - 30539
2021-12-09 11:05:23.833 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:356: BBB48String 0:  Determined range needed 30540 - 30839
2021-12-09 11:05:23.833 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:356: BBB48String 0:  Determined range needed 30840 - 31139
2021-12-09 11:05:23.833 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:356: BBB48String 0:  Determined range needed 31140 - 31439
2021-12-09 11:05:23.833 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:356: BBB48String 0:  Determined range needed 31440 - 31739
2021-12-09 11:05:23.834 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:356: BBB48String 0:  Determined range needed 31740 - 32039
2021-12-09 11:05:23.834 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:356: BBB48String 0:  Determined range needed 32040 - 32339
2021-12-09 11:05:23.834 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:356: BBB48String 0:  Determined range needed 32340 - 32639
2021-12-09 11:05:23.834 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:356: BBB48String 0:  Determined range needed 32640 - 32939
2021-12-09 11:05:23.835 (5579) [ChannelOut] /opt/fpp/src/channeloutput/channeloutput.cpp:395: Determined range needed 30240 - 32939
```

sslupsky, 2021-12-09:
As a workaround, I will disable automatic upload of the output configuration in xLights.  I won't be needing to update the configuration very often so I can do a manual update and go in a make the change manually until xLights is updated.

I'll leave this issue open until xLights is updated.

computergeek1507, 2021-12-09:
I could make a new epprom that has the same id as well

sslupsky, 2021-12-09:
I do not intend to change the output configuration at this point.  

Unless there is a preference to use the old ID it is probably ok to leave it as it is and I will just make the change manually.

computergeek1507, 2021-12-09:
I think it's is the reason for the pinout version field but the issue I have is I added 4 more ports so xLights has to have multiple configurations too

sslupsky, 2021-12-12:
Yeah, I noticed today that ports 17 to 20 do not appear in the xLights visualizer so I am not able to assign a model to those ports.  Can I manually assign the port in the layout editor?

sslupsky, 2021-12-12:
So I manually edited fpp.xcontroller in the app bundle and added a BBBv2 controller definition to xLights.  This solved the problem and also solved the upload issue.

```
    <Controller Name="BBB16v2">
        <Variant Name="No Expansion" ID="BBB16v2" Base="FPP:FPPStringCapeTTYs">
            <MaxPixelPort>20</MaxPixelPort>
            <MaxSerialPort>2</MaxSerialPort>
            <fpp>1</fpp>
            <fpp1>21,-1</fpp1>
            <fppSerialPort1>ttyS1</fppSerialPort1>
            <fppSerialPort2>ttyS2</fppSerialPort2>
        </Variant>
        <Variant Name="w/Expansion" ID="BBB16v2" Base="FPP:FPPStringCapeTTYs">
            <MaxPixelPort>36</MaxPixelPort>
            <MaxSerialPort>2</MaxSerialPort>
            <fpp>1</fpp>
            <fpp1>21,16</fpp1>
            <fppSerialPort1>ttyS1</fppSerialPort1>
            <fppSerialPort2>ttyS2</fppSerialPort2>
        </Variant>
    </Controller>
```

sslupsky, 2021-12-21:
Created PR in xLights:  https://github.com/smeighan/xLights/pull/3025

closed 2021-12-21 by sslupsky