computergeek1507/PB_16 #12: Missing positions CSV files

MKansagra, 2022-01-05:
I'm following the ordering instructions but I can't seem to find the positions CSV file. 
I'm just ordering the PB16 version and nothing fancy.

Am I looking in the wrong place?

computergeek1507, 2022-01-05:
The PB16 is through hole and cannot be assembled by JLC so no position file is needed

MKansagra, 2022-01-05:
Ahh so if I'm not confusing myself here, I just order the PCB from JLC or whoever else and then use BOM to order the parts myself to assemble it at home?

I was following the ordering instructions which mentioned the positions files so I got confused by that.

Which boards require the positions file if I do need any expansion boards or such in the future?

computergeek1507, 2022-01-05:
Any of the PCBs with "SMD" have surface mount parts that can be assembled by JLC

MKansagra, 2022-01-05:
Wonderful! Thank you!

Can't wait to build the controller and try it.

closed 2022-01-05 by MKansagra