computergeek1507/PB_16 #14: BBB_16_Expansion J1, J2, J3 pins 21-24

MmmDee, 2022-03-18:
This is more of a comment. I realize in your use case, the subject pins can be used as you desire; however, when interfacing a "real" F16V4 16-port Expansion card, Pin 21 is used as CLK (you have it as GND, but not connected) and your pin 23 is used for 3.3v. I don't believe pins 22-24 are used on Dave's board (I have a couple of the expansion boards, none of the controllers). This may be worth a mention somewhere. I appreciate you making this all open source.

computergeek1507, 2022-03-23:
Since FPP cannot drive 4 wire pixels I pulled the clock pin 21 to GND to prevent the buffer from oscillating On the F16v4 I have, pin 23 has v3.3 connected to it. On the v4 Expansion pin 23 went nowhere so I assume it was for future proofing if needed.

MmmDee, 2022-03-23:
Got it, close the issue. Didn't see a way to PM/email you aside from the Issues section. I don't have an F16V4 controller (that's what's driving me to BBB options) so I was unsure what was on p22-24 because, as you say, there's no connection on the F16V4 expansion board itself.

closed 2022-03-23 by MmmDee