computergeek1507/PB_16 #16: BBB16 Expansion

Strawberry1997, 2022-04-16:
While testing the outputs I noticed that port 3, 11, 12, 16, 34, doesn't have any pixel data. I've tried it on 3 different BBB16 expansion controllers to rule out any soldering issues.
test was using BBB16 expansion > Diff Expansion > Receiver_Out_SMD > pixels strands.

Also tested the BBB16 expansion > 16_expansion SMD,  some ports work fine, some no data, some flickering.

computergeek1507, 2022-04-16:
What eeprom file did you use?

Strawberry1997, 2022-04-16:
I haven't used any eeprom file.Sent from my T-Mobile 4G LTE Device

Strawberry1997, 2022-04-17:
I've tried the BBB16 and BBB16v2 eeprom files with no JOY. seems this board needs it's unique file.

computergeek1507, 2022-04-17:
I haven't generator one for it yet, I'll try to do it tonight.

computergeek1507, 2022-04-18:
Unzip this file and try this eeprom

[BBB16-Exp-eeprom.zip](https://github.com/computergeek1507/PB_16/files/8502943/BBB16-Exp-eeprom.zip)

Strawberry1997, 2022-04-19:
I've uploaded eeprom, it didn't seem too help.
Which controller> Vendor > model > variant, should I be using in Xlights?

Strawberry1997, 2022-04-19:
The EEPROM did allow for 48 stings in multiple configurations.
The buttons for the oled control seem to not function. I'm not sure how to configure them in the gpio section.
The voltage is reading vin1 27.2v and vin2 23.9. I'm only sending 5v from a dedicated PS.

computergeek1507, 2022-04-21:
there is no ADC circuits, I will have to fix that

Strawberry1997, 2022-05-03:
Have you had a chance to look at the ADA circuits? is this the cause of the flicker on the 16 expansion SMD board?

closed 2022-06-22 by Strawberry1997