computergeek1507/PB_16 #17: PB 16 SMD CPL

tw0924, 2022-04-20:
I am having issues importing the CPL file to JLCPCB? Any advice on what im doing wrong?

computergeek1507, 2022-04-21:
What is the specific board or file?

tw0924, 2022-04-23:
My apologies, it’s is the PB16 V2 position file. It was giving me a error when trying to upload it

mrpburke, 2022-05-01:
I’ve found a couple discrepancies in the BOM and CPL files and the interactive BOM for this board. I just finished up an order so I can submit a pull request tomorrow with the updated files. Some part numbers are different on JLCPCB, SW4 & SW5 we’re missing from SMD BOM, rotation is off on a few components, and the JLCPCB format is slightly different that the provided csv files.

tw0924, 2022-05-02:
Thank you sir. Just trying to plan ahead for this Christmas! Thank you for all your work into this board and making it open source.