computergeek1507/PB_16 #19: Parts issue? BBB_16_SMD

Crezzie, 2022-05-01:
Hi im building a cart in digi-key to have a go at building my own BBB 16. Im having trouble with one item on your list being "not found" on their website. In the BBB_16_BOM file part no 29 Digi-key reference: 2571-CN34J330CTTR-ND is the only one i cant find are you able to help or update the part number? Im in australia if this helps.

Regards luke

Strawberry1997, 2022-05-01:
I think you have the wrong part #. I don't see this part in the BOM file. It appears you are looking for the 33 ohm resistor network.

mrpburke, 2022-05-01:
> 

It's in the BBB 16 SMD BOM. The LCSC part # that's in the BOM is available in JLCPCB and appears to be the one you want (0603WAF330JT5E).

Crezzie, 2022-05-01:
beautiful found it guys thanks a bunch. also 1 more thing is there an alternative to the part on line 34 Part number for digi-key 811-2196-5-ND. the lead time is crazy and literally out of stock EVERYWHERE its the dc dc 5v converter. 1 site i went to said 52 week lead time.

mrpburke, 2022-05-01:
> beautiful found it guys thanks a bunch. also 1 more thing is there an alternative to the part on line 34 Part number for digi-key 811-2196-5-ND. the lead time is crazy and literally out of stock EVERYWHERE its the dc dc 5v converter. 1 site i went to said 52 week lead time.

I believe 102-P7805-2000-S-ND will also work.

computergeek1507, 2022-05-01:
The regulars have been out of stock since 2020, I have been working on a design with a built-in switcher chip, but they can be hit and miss as well.

closed 2022-05-02 by computergeek1507