computergeek1507/PB_16 #34: BBB_16_SMD U5 Part error

processor83, 2022-07-22:
Just to let you know that this part is not correctly connected.

for users dont order this without changes!

processor83, 2022-07-22:
Also just saw that OLED footprint should not be there in gerber files i understand its there for 3D rendering!

bcarriveau, 2022-07-22:
Not correctly connected in what way, orientation, trace not connected, routed wrong?

processor83, 2022-07-22:
footprint has changed and traces are not connected.  also OLED mounting holes should be removed im working on it to fix those but im not very goof with github to ask a pull of those fixes

processor83, 2022-07-22:
[image]
here is the oled part

computergeek1507, 2022-07-22:
Ill fix it, I was consolidating libraries and might have broke it

processor83, 2022-07-22:
Thank you!

I now also see the bleaglebone silk is showed  but im sure you will see it also!

computergeek1507, 2022-07-23:
This should be fixed now

closed 2022-07-23 by computergeek1507