computergeek1507/PB_16 #36: BBB16 v2

Kosh42, 2022-11-06:
When uploading the binary [here](https://github.com/computergeek1507/PB_16/blob/master/BBB_16/BBB16v2-220513130004-eeprom.bin) the Cape Info in FPP reads as PBv2 instead of BBB. Is this correct?

Kosh42, 2022-11-06:
FYI - My cape is fairly old. PCB says BBB 16 v2. It has the OLED sticking out of the top and the BBB on top rather than underneath. Do I need the V2 or un-numbered version?

bcarriveau, 2022-11-06:
I installed the 220513130004 and it does in fact change it to a pb16v2 
[image]

but my pixel string screen shows the correct cape type, so maybe its just the cape info that is wrong. (this board was previously flashed with firmware from before the july firmware update.) So if that cape type is right it should work just fine! 

but if you want or the one you have does not work on the cape here is the older firmware that you may need a fpp license for or atleast have to email them to get a license or something.
pre 6.x firmware for bbb16v2
[BBB16v2-eeprom.zip](https://github.com/computergeek1507/PB_16/files/9946955/BBB16v2-eeprom.zip)

[image]

computergeek1507, 2022-11-06:
The BBBv2 is incorrect, I have emailed Chris multiple times over the last month to generate me a new one with no response. For the BBB16 with the BBB on the top you need the V1 eeprom file.

bcarriveau, 2022-11-06:
v1 file?
[BBB16-eeprom.zip](https://github.com/computergeek1507/PB_16/files/9946977/BBB16-eeprom.zip)

computergeek1507, 2022-11-06:
Think so

computergeek1507, 2023-04-10:
v2 EEPROM was fix with 8c6638d3b7af76e1e375479646f170d5cb0bfdbd

closed 2023-04-10 by computergeek1507