computergeek1507/PB_16 #38: Need some tips and explanations how to write a code

stefancvetkovic, 2022-12-10:
And this is my first encounter with FPP. Previously I used WLED with dig-uno, quad stuff. But I think that FPP is more powerful including e1.31/DDP/ protocol standard
Can you please tell me if this board is good? I pick beaglebone for main cpu because of its hardware advantages, and Im wondering what I need of softare/hardware tools in order to flash it properly and make it work for this board? also I want to add seprate esp32 for wifi/ble communication and pairing with custom mobile app that I want to make (something mobile xLights)

kingofkristmas, 2022-12-25:
Lol

stefancvetkovic, 2022-12-27:
> Lol

?

bcarriveau, 2023-01-10:
So this is a Cape for a beagle bone. The beagle is not only a FPP device , FPP is also the firmware for the cape.  This may or may not be a good choice for you without a little background but it's really just soldering and FPP has a well documented setup manual.  

Not sure anyone is here to give tutorials on codeing as this project does not require any.

closed 2023-04-10 by computergeek1507