computergeek1507/PB_16 #40: BBB 16 SMD V2.1 Part Placement

brymitch56, 2023-01-10:
When I upload the gerber, BOM and placement files to JLCPCB, many of the through hole parts are not placed correctly. Maybe this isn't an issue, as I've read that they don't do through hole anyway. Picture of their render attached with the components circled that look wrong to me.
[image]

bcarriveau, 2023-01-10:
Select the option to have jlc review it on the board selection options page and they will just adjust it for you.

brymitch56, 2023-01-11:
After playing around with the layout render, I figured out you can rotate and reposition parts within the GUI before adding to your cart. Though it's a bit of a pain, it looks like it should work. I've also found the same issue with the Receiver_Out_SMD, and I expect most other SMD boards.

closed 2023-01-28 by brymitch56