computergeek1507/PB_16 #43: BBB 16v2 OLED power and ground

tika911, 2023-01-26:
Let me start of by saying thanks for sharing this and that I have zero EE training.  I am just a hobbyist and am most likely doing something wrong but I could not get the OLED working.  Everything else is working great.  I started troubleshooting and I did not have power or ground at the OLED header.  Once I figured that out, I was able to find a good source of 3.3v on the pads located directly in front of the oled header.  I wired a couple of jumpers from these pads to the power and ground of the header and got the oled to work.

I had to use a TSR 1.5-2450E voltage converter in place of the 811-2196-5-ND due to availability.  I do not know if that is the reason for issue I am having.

computergeek1507, 2023-01-26:
This is the purpose of these jumper pads under the OLED. Some OLEDS have +3.3 and GND on different pins, I added these jumpers to allow the user to configure the correct pins. You jump them with a blob of solder.
[image]

brymitch56, 2023-01-26:
Are you using the SMD BBB version? There are solder pads on the board that you need to jumper with solder depending on the pin configuration of your OLED board. One for power, one for ground. The pins on some OLEDs are reversed, but instead of jumper pins, the SMD board has pads that need to be bridged with solder. They should be under where the OLED mounts, on the top of the board.

brymitch56, 2023-01-26:
[image]

tika911, 2023-01-26:
Thanks that makes sense.

computergeek1507, 2023-01-26:
On newer designs I have been adding a cutable bridge on the pads I normally use but I didn't add it to this one yet

closed 2023-01-26 by tika911