computergeek1507/PB_16 #46: BBB has no power, board has

stefancvetkovic, 2023-02-22:
Any hint regarding that my BBB does not have power? also display is dead.
I put 5V to the left screws only. I tried to hold power btn, but nothing happens.
Thanks

bcarriveau, 2023-02-22:
The board needs to be powered at the bottom right to supply power to the BBB

stefancvetkovic, 2023-02-22:
But still, I dont have power on my BBB. Im using 5v DC, I can see green light on D6 diode, and I put 5A fuse on that spot. but BBB does not emitting any blue lamps.

computergeek1507, 2023-02-22:
You need to set the J23 jumper to 5V

computergeek1507, 2023-02-22:
And set J28 to the VIN position i.e down

computergeek1507, 2023-02-22:
its a different PCB but if you look at this PDF find the "Test Power" section and verify your voltages. I would do it with the BBB not connected until you know it right.

https://github.com/computergeek1507/PB_16/blob/master/PB_16v1/FPP%20PB16%20Pixel%20Controller%20Manual%20and%20Assembly%20Guide.pdf

stefancvetkovic, 2023-02-22:
Thanks, now I have the power on my BBB!
But still Icant access it over BBB's IP address.
When I disconnect BBB and put it the same UTP cable and its own supply, then it works, I can found on that IP address. Do you maybe know why is that? thanks again

computergeek1507, 2023-02-22:
The OLED will not turn on until you flash the EEprom or set it up in FPP. You also have to solder the jumpers under the OLED to the correct pinout of your OLED

stefancvetkovic, 2023-02-22:
Thanks @computergeek1507  can you point me out with which firmware should I need to flash eeprom? I found it on fpp=> about=> flash
But still it should be presenting fpp welcome screen even if connected to the board, but its not.

computergeek1507, 2023-02-22:
If running FPP 6 you should flash from the cape menu option. It needs to be the V2 version

stefancvetkovic, 2023-02-22:
I found this: https://github.com/computergeek1507/PB_16/blob/master/PB_16v2_SMD/PB16v2-220513130003-eeprom.bin
Also I will first update fpp to to version 6

computergeek1507, 2023-02-22:
it was moved to Help->Cape Info in FPP 6, The file you found in the repo had an error, I uploaded a new one here:

https://github.com/computergeek1507/PB_16/blob/master/BBB_16_SMD/BBB16v2-221118135219-eeprom.bin