computergeek1507/PB_16 #47: BBB_16 EEPROM

joelhauxwell, 2023-03-23:
Hello, 

First of all, thanks for making all of this great stuff available!

I have decided to build some BBB_16 boards to control my 2023 Christmas display and have some PCBs on their way from JLCPCB.

Im struggling to source the exact thru-hole EEPROM from the BOM - are you able to suggest an in-stock alternative?

Thanks,

Joel.

GDWoody, 2023-03-23:
Hi,

Try this from LCSC part number C411624 this is what I used for my boards.

Also, I have made my own open hardware customized versions of Scott's BBB-16v2 SMD free for anyone to try.

https://github.com/GDWoody/Pixel-Controllers

joelhauxwell, 2023-03-23:
@GDWoody thanks for the link, much appreciated 

Also, thanks for sharing your projects - your bbb_all_diff is a great fit for what I need...I feel another JLCPCB order coming on 😊.

computergeek1507, 2023-03-23:
The EEPROM file is 4KB so any EEPROM 4KB or bigger should work. "24LC32A-I/P-ND" or "25LC640-I/P-ND" are two I found doing a quick search. I haven't tested them, so order at your own risk.