computergeek1507/PB_16 #50: Request for help with Pi_Zero_24_Diff

Gooselight, 2024-02-18:
Hi Scott

Firstly, thank you for making all this wonderful material available and congratulations on your dedication to making our learning easier.

My name is Henrique, my nickname is Gooselight, I'm from Brazil and due to the difficulty in purchasing boards to use with BeagleBone Black here where we live, I decided to build some BBB_16 and Pi_Zero_24_Diff boards using their projects as a base.

The BBB_16 boards are working perfectly with version 5.5 of Fpp, but the Pi_Zero_24_Diff boards I'm trying to use with this 5.5 version as well, but unfortunately I'm not having success, I tried using the EEprom version: Pi_Zero_24_Diff-eeprom but it still didn't work.
Could you tell me if it is possible to use this scheme to work with this version of Fpp and this eeprom, or will it only work with versions above Fpp 6 or newer that need a signed eeprom. Can you help me with this?

Thank you once again for your material and help.

Henry B Garcia
Maringá PR Brazil

[image]
My board.

computergeek1507, 2024-02-18:
The DPI pixel output used by the pi based boards need FPP 6 or newer

Gooselight, 2024-02-18:
Thank you for quicly answer

Enviado do meu Samsung Mobile da Claro
Obter o Outlook para Android<https://aka.ms/AAb9ysg>

closed 2024-06-28 by computergeek1507